bambulab/BambuStudio #5420: PETG-HF appears as ABS until Sychronize with AMS is done

mgaulin, 2024-11-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I have the latest firmware loaded in the X1. I am running the latest Studio. I have loaded the latest filament resource pack.

### Printer

X1 Carbon

### How to reproduce

1. Observe that PET-HF black is loaded into the AMS in slot #2, and that I have been printing with it already today.
2. Open Studio
3. Go to Prepare pane
4. Note "ABS (black)" appears to be loaded in AMS slot #2; this is not the correct filament.
5. Hit Synchronize filaments with AMS
6. I get the error "No AMS filaments.. Please select a printer in Device page". Switching to the Device pane and then returning to Prepare will fix this error. Hitting Sync again may give the notice "Already did a synchronization, do you want to  sync only changes or resynch all?". Note that "ABS" is still listed in slot #2.  Hit Resync.
7. Observe that slot #2 now shows PETG-HF (black); which is correct

### Actual results

On startup the wrong filament is shown (ABS vs PETG HF)

### Expected results

On startup the correct filament should be shown. (All other filaments are correct)

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-29:
I’ve summarized your issue as follows. Please let me know if this is correct:  

- **ABS in Slot 2:** This phase occurred before you described the entire issue. You successfully printed using ABS in this slot.  
- **Replacing ABS with PETG HF in Slot 2:** At this point, you closed Studio (note that Studio automatically records the current preset status, including the filament type, when exiting. This means the next time you open Studio, Slot 2 will still default to ABS). After replacing the filament, the X1C correctly recognized PETG HF. However, when reopening Studio, Slot 2 still displayed ABS, as mentioned earlier.  
- **Synchronizing Filament Information:** At this stage, you attempted to sync AMS information. However, it’s clear Studio required you to select the device first. After choosing the X1C and returning to the preparation interface, the synchronization took effect this time.

It seems not a bug in my description, I think the only problem in the whole process is user may be confused by why he has to switch to Device tab to set the X1C before a right sync.It is a problem especially for those who only has one printer.

mgaulin, 2024-11-29:
Negative, JohnSmithCYM, I never had ABS actually loaded in the AMS at any time; Studio is erroneously displaying "ABS".  At no time did I change what is physically loaded in the AMS.  This is entirely an issue with Studio appearing to not remember what it learned from the last Sync with the AMS (or misinterpreting what it stored last time the app exited, and deciding the PETG-HF is not that, ad must be ABS instead.