bambulab/BambuStudio #5421: Snug support style generates bugged slicing

ArnieO, 2024-11-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor: Intel(R) Xeon(R) E-2126G CPU @ 3.30GHz   3.31 GHz
RAM: 32 GB

### Printer

Bambu Lab A1

### How to reproduce

Import attached ´project file, which has painted support on positions that need support.
Slice it with these support settings:

- Enable support: Enabled
- Type: normal(manual)
- Style: Default
- On build plate only: Disabled
- Remove small overhangs: Enabled

When Style is changed to Snug, the slicing is correct.

### Actual results

Some of the supports penetrate through part and down to build plate:
[image]

### Expected results

Supports should be only between parts, like on the adjacent row:
[image]

Sliced with `Style: Snug`:
[image]

### Project file & Debug log uploads

[Pow-K+ holders 32pcs.zip](https://github.com/user-attachments/files/17937267/Pow-K%2B.holders.32pcs.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-11-28:
This is how normal support (grid style) works, not a bug. The two pillars are considered a whole connected component; it's just that the connection is too thin to be extruded.
[image]

ArnieO, 2024-11-28:
The bug changes with selected layer height profile (System presets):

- 0.08mm High Quality and Extra Fine
    0.12mm Fine and High Quality
    0.16 mm High Quality and Optimal:
    One inter-object support is larger than the other, but no foot underneath towards build plate:
    [image]
- 0.20mm Standard and Strength:
    One inter-object support is larger than the other, a foot is added underneath towards build plate:
    [image]
- 0.24mm Draft:
    One inter-object support is larger than the other, a foot is added underneath towards build plate. The inter-object support and the foot are different compared to 0.20mm layer height:
        [image]
        [image]
- 0.28mm Extra Draft:
    The result is as expected for correct slicing: Inter-object supports are identical, no foot added underneath towards build plate.
    [image]

ArnieO, 2024-11-28:
In the last tests, I removed one letter from the model, in the area where the foot gets generated - to check if that was the reason. 
STL file for this configuration with one single row, for testing: 
[Holdere.zip](https://github.com/user-attachments/files/17947697/Holdere.zip)

ArnieO, 2024-11-28:
I tried slicing for 0.6 mm nozzle / 0.3mm Standard - and the problem moved to the adjacent support:
[image]

ArnieO, 2024-12-06:
When studying the slicer result, we see there is support material inside the part, I think this confirms that it is a bug:
[image]

ArnieO, 2025-01-21:
@tangketan : Any update on this? Has Bambu Lab recognized it as a bug?