bambulab/BambuStudio #5422: AMS to printer connection

cseelow, 2024-11-27:
### Bambu Studio Version

Current version

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Printer

X1C

### How to reproduce

Open a model from maker world. Their filament isn't going to match my AMS loaded filaments so I hit the sync AMS button. The message tells me no filaments, no AMS. I go to device page, click nothing, go back to the prepare tab and hit the snyc AMS and it loads the printer and the filaments.

### Actual results

doesn't load printer AMS or filaments

### Expected results

supposed to load AMS filaments when hit snyc

### Project file & Debug log uploads

[PicatinnyMP7Grip-Hinge-Pic-A.zip](https://github.com/user-attachments/files/17937850/PicatinnyMP7Grip-Hinge-Pic-A.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-29:
Like what I mentioned in https://github.com/bambulab/BambuStudio/issues/5420, it seems that the key problem is most people will sync after they load the 3mf or open a link from makerworld after the open Studio, they didn't switch to Device tab to choose a printer or make sure the right printer they are using, it is a problem especially for those who only have one printer.

mgaulin, 2025-01-02:
Hi. I am the reporter for #5420 (mentioned above by JohnSmithCYM), and yes, I only have one printer and switching to the device tab does sort of allow the filaments in AMS slots to appear correct in Studio, but it seems weird to me that the other three AMS filament slots are always correct, and all four of them were always were correct before I loaded the new PETG-HF in one slot; it is as if the "last known state" of the AMS is retrieved at startup and displayed correctly, except for that one (new-ish) filament type, which it maybe doesn't recognize correctly from the loaded "last known state". (I have not looked at any of this code; I am just describing what it feels like.) -Thanks