bambulab/BambuStudio #5424: Text orientation resets

flatpick, 2024-11-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Window 10

### Printer

Bambu labs X1C

### How to reproduce

1. Create or insert an object.
2. Select the object and press "T" to add text
3. Once text is added, change text orientation.  In my case I moved the text to stack in line with a drink clip (see image)
4. copy and paste the object
5. Select the pasted object
6. In the "Object" menu, right click on the "text shape" and change the text in the "Input Text" field
7. The text orientation should change/reset as soon as anything is changed in the text field
8. 
[image]

### Actual results

When editing text on an object that has been moved, rotated, or otherwise reoriented, as soon as anything in the "Input Text" field is changed the text on the plate reorients

### Expected results

The text position should not reorient every time text is changed, orientation should remain static unless manually edited

### Project file & Debug log uploads

[Name tag trouble file.zip](https://github.com/user-attachments/files/17940706/Name.tag.trouble.file.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-11-28:
Hello, we are considering maintaining the original position and posture while continuing to edit the text~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00