bambulab/BambuStudio #5425: near inaudible MacBook *hardware* clicking noise only in very specific interactions (detected by cat)

reneroth, 2024-11-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1.1 (24B91)

### Additional system information

MacBook Pro 16-inch, Nov 2023, M3 Pro

### Printer

Bambu P1S

### How to reproduce

Please stay with me, I swear there is something going on here.

1. Get your MacBook, sit down, have your cat position herself on your lap and fall asleep in a way that her head touches the hardware. This is hard to get right, but essential. I have attached a diagram below outlining common mistakes.
2. Wait until it is _really_ quiet at night. Completely turn off the audio on your MacBook.
3. Open `Bambu Studio`.
4. Load any project or model, slice, go to `Preview` tab. (It does not work without a model in there.)
5. Using the MacBook's trackpad, move your cursor over the 3D preview.
6. Notice that your cat's ears are twitching whenever your cursor moves over the 3D render in the `Preview` tab. Notice that the ears are not twitching on any other interaction.
7. Seriously. Her ears twitch, but only in that very specific instance. Double check the sound of your machine is truly turned off. Maybe enjoy for a bit that you can make your cat's ear twitch by moving your cursor.
8. Hold your breath, put your ear really close to your MacBook, try moving the cursor again.
9. Hear a very faint clicking from inside your machine, like the old sound of HDDs back in the day. It's _very_ quiet, maybe below the audible threshold for many people?
10. Crank up the volume and try again to confirm this is not a sound being played. This is the hardware producing clicky sounds.

**TLDR: MacBook makes hardware noises when moving cursor over active 3D Preview.**

My only viable theory is that there is some bugged code trying to produce tactile feedback with too little strength, so the vibration motor inside the touchpad does not start and only makes this faint clicking noise.  
It's probably inconsequential in the grand scheme of things, but you know what they say about curiosity and cats. Who knows, maybe it's related to some other bug?

[image]

### Actual results

There is a near-inaudible hardware noise coming from the MacBook.

### Expected results

There should not be a near-inaudible hardware noise coming from the MacBook.

### Project file & Debug log uploads

Use any model to reproduce bug.

I can email logfiles to devs privately if needed, they seem to be full of personal data. Arranging cat photos is more fun than sanitizing log files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-11-28:
Hello, may I ask if our software interface is working properly？

reneroth, 2024-11-28:
> Hello, may I ask if our software interface is working properly？

Everything works perfectly fine, functionality wise and visually!

reneroth, 2024-11-28:
I debugged this a bit further, I don't think it's haptics. I actually think it is very subtle coil whine.

It seems to be caused by Bambu Studio doing something with a lot of CPU and GPU load whenever the cursor moves. The load _immediately_ goes down when I stop the cursor.  
When cursor is idle, CPU and GPU usage are at around 1% each. As soon as the cursor moves over the render preview window, it goes to over 40% each. Nothing is visibly changing in the app or UI. Not sure what Bambu is doing here.
You can see it in the CPU and GPU history:

[image]

I just went back and checked: PrusaSlicer does not have a `Preview` panel, but is even worse (goes to 80% CPU and 60% GPU when moving cursor). OrcaSlicer has the exactly same behavior. Just guessing: are there some really intense calculations done whenever the cursor moves, maybe completely unthrottled, so a higher refresh rate might make it worse?

It's not a dealbreaker, as the app never becomes unresponsive or sluggish, it's mostly my curiosity bringing me back to this.

Haidiye00, 2024-11-29:
Hello, when the mouse enters the 3D display interface, there will be some updates to the 3D display interface, which will increase the load on the CPU and GPU~

reneroth, 2024-11-29:
> updates to the 3D display interface

If it's always supposed to happen we can close the issue! I was just surprised because it seems like a lot of calculations when the cursor moves, but nothing is happening visually.
Performance is fine though, so probably nothing worth optimizing for now?

Thank you for this software, it finally made printing fun again!!

Haidiye00, 2024-11-29:
Hello, I'm sorry but we won't optimize it for now. Although the 3D interface may not change much during mouse sliding, in reality, the software will detect whether an object has been hit during mouse sliding, and the 3D interface may need to make changes~

closed 2024-11-29 by reneroth

JohnSmithCYM, 2024-11-29:
Very lovely cat and I'm sure it is a good tester 🐱 🐱 🐱 !