bambulab/BambuStudio #5430: Preview Layers incorrect

timothygschmidt, 2024-11-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 64-bit

### Additional system information

T1-13900LS-RTX470; 64 GB

### Printer

Bambu Lab A1

### How to reproduce

Prepare using Bambu Lab A1 0.4 Nozzle, 0.12 Fine @BBLA1 and slice the model. 

### Actual results

Total layers = 787,  as you arrow down the layer count jumps up to 1,000.

### Expected results

The layer number should decrement downward from 787 and not jump back up.

https://github.com/user-attachments/assets/bc3ef3df-c0cb-4663-9ad5-56e8c76e0272

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17950644/log.zip)
[Cogsworth funky.zip](https://github.com/user-attachments/files/17950649/Cogsworth.funky.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnSmithCYM, 2024-11-29:
[image]
It's wired, I used your 3mf on V1.10.1.50 like the gif above, no problem shows up and can you consistently reproduce this bug now？

timothygschmidt, 2024-11-29:
I rebooted and have not seen it since.  It may have been because I changed from a fine 0.2mm head 0.10mm slice earlier to a 0.4mm 0.12 mm slice and then printed. :(

 
 

20241129175825_rec_.gif (view on web)