bambulab/BambuStudio #5431: Brim Ears do not print - dependant on unknown value in machine configuration

IRIS-Kane, 2024-11-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Creality Ender 5 Plus

### How to reproduce

1. Open Bambu Studio
2. Drag STL onto build plate
3. Set **Others > Brim** Type to **Painted** (tried both Global and Object settings)
4. Select object and use **Brim Ears [L]** tool to paint brim
5. Select **Slice Plate**
[image]
[image]
[image]

### Actual results

Model is sliced with no brim.

[image]
[image]
[image]

### Expected results

Model should be sliced with brim ears.

### Project file & Debug log uploads

After some troubleshooting I've found **Brim Ears will generate** if the machine is set to any of the preloaded Bambu printers, but will not generate on the Creality Ender 5 Plus, or custom machine configuration. I am not sure what setting the brim ears are dependent on and the Wiki page does not specify anything machine specific.

[ProjectFile.zip](https://github.com/user-attachments/files/17952691/ProjectFile.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

IRIS-Kane, 2024-11-28:
I just confirmed brim ears are working on the **Prusa MK3S 0.4 nozzle** system preset, but not the **Creality Ender-5 Plus 0.4 nozzle** system preset ¯\_(ツ)_/¯

MackBambu, 2024-11-29:
The issue can be confirmed, and it's not just the brim ears. other brim types are also not generated. This problem could not be reproduced on other third-party printers. I suspect there might be some incorrect configuration with the Ender 5 Plus printer.

MackBambu, 2024-11-29:
Confirmed it's a bug.
Temporary workaround: first set the brim type to "Outer brim only," then set any value for the brim width, and finally switch the brim type back to "Painted".

xigasa, 2025-01-29:
> Confirmed it's a bug. Temporary workaround: first set the brim type to "Outer brim only," then set any value for the brim width, and finally switch the brim type back to "Painted".

Can confirm this fixed my issue.

sepp32, 2025-03-25:
> Confirmed it's a bug. Temporary workaround: first set the brim type to "Outer brim only," then set any value for the brim width, and finally switch the brim type back to "Painted".

This fixed my issue too. Stilll a bug in Bambu Studio

MackBambu, 2025-04-02:
Is the issue still present in the latest version ?

IRIS-Kane, 2025-04-02:
I just tested on 2.0.1.50 and it does appear fixed. BambuStudio generated the brim ears without needing to do the workaround to set the Brim Width greater than 0 first. Thanks for resolving it!

closed 2025-04-02 by IRIS-Kane