bambulab/BambuStudio #5432: Studio cannot slice and crashing

dimakad, 2024-11-29:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11 pro

### Additional system information

i9-13900K, 64gb RAM.

### Printer

Bambu lab X1CC

### How to reproduce

Open mode, slice.
I've tried few differed Sword-kind of models from MakerWorld, and they all cannot be sliced in nether Classic not Arachne modes
https://makerworld.com/en/models/114630?from=search#profileId-151025
Also attaching STL file from another web-location (Tanjiro_Collapsing_Katana1), which cannot be sliced as well.

### Actual results

Studio Crash, closed

### Expected results

Model is sliced

### Project file & Debug log uploads

[Reverse_Blade_Sword_X1P1A12(4).zip](https://github.com/user-attachments/files/17952954/Reverse_Blade_Sword_X1P1A12.4.zip)
[Tanjiro_Collapsing_Katana1.zip](https://github.com/user-attachments/files/17952967/Tanjiro_Collapsing_Katana1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

JohnSmithCYM, 2024-11-29:
[image]
[image]
[image]

Everything seems work for me in V1.10.1.50 both in arachne and classic...Can you offer your crash log and dmp file?

dimakad, 2024-11-29:
[crash_Fri_Nov_29_20_52_46_2.zip](https://github.com/user-attachments/files/17962102/crash_Fri_Nov_29_20_52_46_2.zip)
Here are files from the log folder after slicing has crashed the studio.

Found how to over step this problem. The solution is not convenient as I need to make it each time Bambu Studio is starting, but at least it is working.
Go to task manager -> right click on Bambu Studio process -> Turn on Efficiency mode

QingZhangBambu, 2024-12-02:
@dimakad 
Could you try the following steps, it may works，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing