bambulab/BambuStudio #5433: Ironing does not apply to single-layer thick part surface

briansemrau, 2024-11-29:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Add primitive, scale height to 0.2mm
2. Set Ironing Type to Top Surfaces
3. Slice

Looking at the slicing results, it appears that ironing only happens atop lines labeled "Top surface", but single layer parts only have "Bottom surface" type lines.

### Actual results

[image]

### Expected results

See results with 0.4 (2 layers) part thickness
[image]

### Project file & Debug log uploads

[ironing_bug_reprod.zip](https://github.com/user-attachments/files/17953026/ironing_bug_reprod.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2025-04-02:
good question, i will check it

QingZhangBambu, 2025-04-03:
[image] I think that's what you want