bambulab/BambuStudio #5435: Script Text partially prints

customputing, 2024-11-29:
### Bambu Studio Version

1.9.7.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon with AMS

### How to reproduce

1. Add Text to a model
2. Choose a script text such as Edwardian Script ITC
3. See thinner lines in the text block do not print 
4. I tried detect thin walls and it seemed to improve some but not very much
Note: I also tried fonts between the ranges of 12 and 16 with the same results

### Actual results

[image]
It does not print the thinner parts of the text.

### Expected results

The thinner parts of the Text object should still print with the chosen color

### Project file & Debug log uploads

The 3mf file zipped is 29MB so I can not attach it to this.
[image]

Is there a way to just export the configuration and have the necessary files in a smaller file?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-12-02:
The same issue https://github.com/bambulab/BambuStudio/issues/4965
HI [customputing](https://github.com/customputing) ：Please try following the steps below，then slice again.
[image]

customputing, 2024-12-03:
That did do a little bit better but it still is printing some of it in black instead of silver. It's even raised the .25 as black as you can see here.

image.png (view on web)<https://github.com/user-attachments/assets/858d938d-580b-45ea-934f-e30ed9605e04>

customputing, 2024-12-03:
Edit, it is not raised where it's not silver, it's indented a little.

QingZhangBambu, 2024-12-04:
@customputing Sorry, it's too small to print, make it bigger