bambulab/BambuStudio #5436: a special character in a user defined filaments id crashes network plugin

eingemaischt, 2024-11-29:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu 24.04 and Win 11

### Printer

A1

### How to reproduce

I don't know how this special character came into the filaments id.

### Actual results

The Bambu Studio can't connect to A1 - neither by cloud nor in lan mode. Bambu Android Apps works....

### Expected results

Connection possible.

### Project file & Debug log uploads

Have a closer look into https://forum.bambulab.com/t/connecting-in-bambu-studio/116445/13

My bambu studio did not connect for the last three days so I wanted to use an open source client. But that crashed, too. Unless the bambu studio network plugin it gave out an UTF-8-coding error. Have a look at this filament description that was set to the external spool holder:
```
"vt_tray":{"id":"254","tag_uid":"0000000000000000","tray_id_name":"",
"tray_info_idx":"Pde2\xea58c",
"tray_type":"TPU","tray_sub_brands":"","tray_color":"FFFFFFFF","tray_weight":"0","tray_diameter":"0.00","tray_temp":"0","tray_time":"0","bed_temp_type":"0","bed_temp":"0","nozzle_temp_max":"250","nozzle_temp_min":"200","xcam_info":"000000000000000000000000","tray_uuid":"00000000000000000000000000000000","remain":0,"k":0.019999999552965164,"n":1,"cali_idx":0}
```

The solution after seeing this was simple: I went to the printer and assigned a generic pla filament to the external spool holder - and it's connecting again.

Interesting side fact: Even a factory reset before or coupling another account did not help. I don't know if the printer keeps the filament information in a non-volatile area or if it resynced it right after factore reset with data from the cloud after connecting my account....

### Checklist of files to include

- [ ] Log file
- [ ] Project file

eingemaischt, 2024-11-29:
Mentioned in https://forum.bambulab.com/t/connecting-in-bambu-studio/116445/14

walterwongbbl, 2024-12-12:
Hi. @eingemaischt 
Can you provide printer logs or screenshots? I need to know what filament it is.

eingemaischt, 2024-12-12:
Hi,

can you have a look at Ticket _EU_US241128462001_APP_ANDROID_5641 ?

eValker, 2024-12-25:
Hi,

I can confirm the issue described here. This morning I was printing things on the printer, and everything was working fine, but suddenly in the evening, it was no longer possible to connect to the printer via BambuStudio. I tried connecting from two different laptops, and the result was the same—on one, the connection couldn't be established, and on the other, the connection seemed to exist, but the printer status didn't update. Restarting the studio, computer, or router didn't help. Interestingly, everything worked fine via the Android app.

I tried removing the printer from the account and adding it again, but the result was still the same—stuck at connection, though without a connection error. Fortunately, I found this thread—indeed, after setting every filament to generic PLA, BambuStudio instantly connected to the printer, and everything started working again.

The strangest part is that I haven't changed the filament configuration for a long time. The names of the custom settings also don't contain any special characters—only letters and numbers. Switching back from generic PLA to custom configurations didn't cause the printer to disconnect, and it now works correctly. I tried checking the logs for any errors, but unfortunately, I couldn't find anything.

Attached is the custom filament configuration I found in the configuration files.

I hope this helps in identifying the issue. If you have any questions, I will do my best to provide additional information :)

[Colorfil PLA Basic @Bambu Lab A1 mini 0.4 nozzle.json](https://github.com/user-attachments/files/18247287/Colorfil.PLA.Basic.%40Bambu.Lab.A1.mini.0.4.nozzle.json)
[Devil Design PLA Basic @Bambu Lab A1 mini 0.4 nozzle.json](https://github.com/user-attachments/files/18247288/Devil.Design.PLA.Basic.%40Bambu.Lab.A1.mini.0.4.nozzle.json)
[SUNLU PLA Silk Multi @Bambu Lab A1 mini 0.4 nozzle.json](https://github.com/user-attachments/files/18247289/SUNLU.PLA.Silk.Multi.%40Bambu.Lab.A1.mini.0.4.nozzle.json)

aotillar, 2025-01-11:
Just had the same issue. I changed Filament and my printer was not working. After removing the custom filament on Bambu studio AND on the Printer, as well as changing the filament on the printer to generic PLA, the Printer was able to connect.