bambulab/BambuStudio #5439: stp files crashing

rgancorz, 2024-11-30:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 pro build 19045.5198 22h2

### Additional system information

Device name	Dell-Evo
Processor	Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz   3.40 GHz
Installed RAM	32.0 GB
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
nvidia rtx 2000 driver 566.03

### Printer

bambu x1c

### How to reproduce

import stp file
preview
use z slider to see slice levels

### Actual results

bambu studio closes

### Expected results

i can view the slices

### Project file & Debug log uploads

[drip pan.zip](https://github.com/user-attachments/files/17966462/drip.pan.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

rgancorz, 2024-11-30:
[bambu-studio.exe.14424.dmp.zip](https://github.com/user-attachments/files/17966479/bambu-studio.exe.14424.dmp.zip)

MackBambu, 2024-12-04:
The TBB library may have issues handling multithreading tasks on older CPUs.

MackBambu, 2025-04-29:
The new version has already switched to a different threading model. Does the issue still persist?