bambulab/BambuStudio #5440: Opening in Bambu Studio, crash Bambu Studio on my Mac Mini M2 Pro

Telecino, 2024-11-30:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1.1 (24B91)

### Additional system information

CPU: Apple M2 Pro
Memory: 16 GB
GPU: Apple M2 Pro

### Printer

Bambu Lab X1 carbon

### How to reproduce

GO to Maker World,
Select any file,
select open in Bambu Studio,
Bambu studio load and crash

### Actual results

[image]

### Expected results

not Crash!

### Project file & Debug log uploads

[Log.zip](https://github.com/user-attachments/files/17966789/Log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-12-02:
Hello @Telecino 

could you help to export the crash report following the guide in this page?
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

thanks

Telecino, 2024-12-02:
Hi Lane Wei,

 

Here’s the requested crash report.

 

I hope this helps,

 

Best regards,

 

Stéphane

 

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

lanewei120, 2024-12-05:
Hi @Telecino 

did you reply by email?
it seems the attachment in email can not be forwarded to github

we still can not find it above

thanks

Telecino, 2024-12-05:
Dear Lane Wei,

Here is the crash report.

Stéphane

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [11208]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.10.01.50 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-02 09:29:51.9796 -0500
OS Version:            macOS 15.1.1 (24B91)
Report Version:        12
Anonymous UUID:        E5B75680-D6C1-D983-31DF-BBE2BA9019E8

Sleep/Wake UUID:       34E22468-2B08-4088-B589-6A03CDA21A54

Time Awake Since Boot: 810000 seconds
Time Since Wake:       437422 seconds

System Integrity Protection: enabled

Crashed Thread:        39

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   BambuStudio [11208]

Application Specific Information:
abort() called

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1983153c8 __semwait_signal + 8
1   libsystem_c.dylib             	       0x1981f4714 nanosleep + 220
2   libsystem_c.dylib             	       0x1981f462c usleep + 68
3   BambuStudio                   	       0x1030594d8 Slic3r::GUI::Plater::import_model_id(wxString) + 1784
4   BambuStudio                   	       0x10302e358 Slic3r::GUI::Plater::priv::on_action_request_model_id(wxCommandEvent&) + 244
5   BambuStudio                   	       0x105401fe0 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
6   BambuStudio                   	       0x105401cc0 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
7   BambuStudio                   	       0x105401ae4 wxEvtHandler::ProcessEvent(wxEvent&) + 56
8   BambuStudio                   	       0x105401934 wxEvtHandler::ProcessPendingEvents() + 236
9   BambuStudio                   	       0x105365cfc wxAppConsoleBase::ProcessPendingEvents() + 112
10  BambuStudio                   	       0x10540a7f0 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
11  CoreFoundation                	       0x198439078 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
12  CoreFoundation                	       0x198438f64 __CFRunLoopDoObservers + 552
13  CoreFoundation                	       0x198438594 __CFRunLoopRun + 788
14  CoreFoundation                	       0x198437bc4 CFRunLoopRunSpecific + 588
15  HIToolbox                     	       0x1a38a7f64 RunCurrentEventLoopInMode + 292
16  HIToolbox                     	       0x1a38adbb4 ReceiveNextEventCommon + 220
17  HIToolbox                     	       0x1a38adeb8 _BlockUntilNextEventMatchingListInModeWithFilter + 76
18  AppKit                        	       0x19bf63a08 _DPSNextEvent + 660
19  AppKit                        	       0x19c8a3e0c -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
20  AppKit                        	       0x19bf56ae0 -[NSApplication run] + 480
21  BambuStudio                   	       0x105332330 wxGUIEventLoop::OSXDoRun() + 112
22  BambuStudio                   	       0x10540aed4 wxCFEventLoop::DoRun() + 40
23  BambuStudio                   	       0x10538a590 wxEventLoopBase::Run() + 132
24  BambuStudio                   	       0x105365590 wxAppConsoleBase::MainLoop() + 120
25  BambuStudio                   	       0x1052f8640 wxApp::OnRun() + 36
26  BambuStudio                   	       0x1053b806c wxEntry(int&, wchar_t**) + 148
27  BambuStudio                   	       0x102f7d42c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 468
[... 720 lines cut ...]
      "deploymentId" : 400000007
    }
  ]
}
}

Model: Mac14,12, BootROM 11881.41.5, proc 10:6:4 processors, 16 GB, SMC 
Graphics: Apple M2 Pro, Apple M2 Pro, Built-In
Display: MXX-A-2-H, 1920 x 1080 (1080p FHD - Full High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Sep 23 2024 18:39:43 version 23.900.1.1.41.51.169 FWID 01-b725314d
IO80211_driverkit-1315.7 "IO80211_driverkit-1315.7" Sep 28 2024 16:04:40
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: 2.4G Receiver
USB Device: HD Pro Webcam C920
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

Telecino, 2024-12-05:
Dear Lane Wei,

 

I added the crash report directly to GitHub.

 

Let me know if you need anything else.

 

Best regards,

 

Stéphane

walterwongbbl, 2024-12-06:
@Telecino 
Did you change the storage location after the file is downloaded?

fahaaaxiki, 2024-12-06:
There is a known issue that changing the storage location of the download into the "/system" path may cause a crash. If you did change your download path, you can change it into a new path that has read-write authority in Preference, which can solve this problem temporary.
[image]

Telecino, 2024-12-10:
> Hello @Telecino
> the default directory is in /Users/user.name/Downloads

Hi Lane Wei,

 

Thank you for the information

 

Have a nice day,