bambulab/BambuStudio #5441: Adding Filament profiles to AMS - FAILS.

NVNDO, 2024-11-30:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambulabs P1S

### How to reproduce

Go to "Device" tab - AMS Spools - and add custom filament presets.

https://github.com/user-attachments/assets/278464c9-94ac-46e6-a948-b5cbb854530d

Some will add and stay - other wont. 

### Actual results

They can be added, at first, but will then vanish when adding a second or third profile (even when just waiting)
It does not happend with all profiles though... so something is very off here.

### Expected results

All filament presets should stay once added to the AMS. 

### Project file & Debug log uploads

[Bambu_log.zip](https://github.com/user-attachments/files/17967286/Bambu_log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

NVNDO, 2024-12-01:
It seem to be related to the color-coding in those filament profiles. ⬛⬜🟧🟪🟨🟩🟥🟦
I tried adding other profiles without color and they work/stay in place. The ones color-coded vanish. There is clearly a bug here.

NVNDO, 2024-12-07:
Hey @walterwongbbl, will this be looked at? I can't get to my AMS set like this...

walterwongbbl, 2024-12-12:
Hi @NVNDO 
This seems to be caused by the lack of synchronization of material information on the printing side.

DanBao-Bambu, 2024-12-12:
This issue should not be related to color settings, but rather to the reset caused by Studio/Orca being connected to the printer simultaneously and one instance not having the custom Filament. If there is no such problem, you can try logging in again on the printer side.

Tartare-Bambu, 2025-08-20:
Thank you for your feedback. Here is our latest update regarding the second and third issues you mentioned:

- **Second issue**: We have identified the cause and will provide a fix in our next release.
- **Third issue**: We regret that a direct “reset to no color” option is not currently available. However, in the default system color picker, “no color” is effectively represented as white.

We sincerely appreciate your understanding and support, and we will continue to improve the experience in future updates :)

NVNDO, 2025-09-15:
@Tartare-Bambu, and yet another encounter. Saving **custom** colors does **NOT** work either. 

I had previously saved different colors from the existing color-palette and they stayed in my Custom color palette. 
But when I add colors which are not part of the existing palette - They do not save.

See here: https://youtu.be/TIjHuwK9xJU