bambulab/BambuStudio #5442: Filament listing no longer alphabetical order

tangcla, 2024-12-01:
### Bambu Studio Version

1.10.1

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1 C

### How to reproduce

1. Drop-down filament list

### Actual results

Expectation is the list of filaments are sorted alphabetically

### Expected results

List seems to be random or unsorted
[image]

[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17967517/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

brianharwell, 2024-12-04:
This is duplicate of https://github.com/bambulab/BambuStudio/issues/5318

DanBao-Bambu, 2024-12-12:
This requirement has been recorded in the requirement list.