bambulab/BambuStudio #5447: Slicer preview is perfect, but after changing to support layer x/y coordinate moves approx 1mm away during printing

IT-Systemhaus, 2024-12-01:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

i7 11-gen, 16 GB, 2TB NVME

### Printer

Bambu P1S with AMS

### How to reproduce

When i print my object with  without using support filament, the print is perfect, but worse to remove the support material. When i change the top layer for support, to get smoother surface, the i get an offset in x/y direction for approx 1mm
[image]
Light picture is printed without support layer. All with PETG HF. Left picture was printed by using support matrial.

### Actual results

When the printing head change the FIRST time from PETG to SUPPORT PLA, this unexpected offset appears and he continues printing with this offset.
[image]

### Expected results

right picture from my upload before.

[Rive Knopf  7.8mm Dezember 2024 v2.zip](https://github.com/user-attachments/files/17970132/Rive.Knopf.7.8mm.Dezember.2024.v2.zip)
I think this problem start with slice level 20. 
[image]
I got no warnings, no problems. I see this result only during printing. No faults in preview.
Logfiles i can upload, if you really need it. I worte this message from another computer, where Bambulab is not running

### Project file & Debug log uploads

[
[Rive Knopf  7.8mm Dezember 2024 v2.zip](https://github.com/user-attachments/files/17970178/Rive.Knopf.7.8mm.Dezember.2024.v2.zip)
[debug_Sun_Dec_01_10_51_51_25008.log.zip](https://github.com/user-attachments/files/17970182/debug_Sun_Dec_01_10_51_51_25008.log.zip)

](url)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-12-02:
It seems a pure printer issue, unrelated to slicing. I'll try to print one and reply back to you.

IT-Systemhaus, 2024-12-06:
Many thanks for printing. I'm happy, thats not a slicing problem. I will try to modify the flushing volume for tower, when the printjob change from PETG to PLA SUPPORT material. I clean the build plate carefully before each printing. Best wishes. Maik