bambulab/BambuStudio #5449: filament sequence is not working

ddavie6990, 2024-12-01:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu X1c

### How to reproduce

[image]
It always prints 7 after 3 and 8. 

### Actual results

will not print filament 7 first as it is set in the image. 

### Expected results

You can change the order of 3 and 8, but it always print 7 after them.

### Project file & Debug log uploads

[Filament Sequence Issue.zip](https://github.com/user-attachments/files/17970765/Filament.Sequence.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-12-04:
HI [ddavie6990](https://github.com/ddavie6990)：Please try again according the gif.It will be 13(green)--3(red)--8(grey)[image]

ThorsReign, 2025-05-06:
This issue of not following filament ordering continues through latest version - using version 2.0.3 you can see in the attached that I have set for layers 2-End to use slot 3 first, but on the layers where slot 3 appears it never prints first.  It's the only slot which does this.

[AMS Purging Strips Three Colors.zip](https://github.com/user-attachments/files/20052015/AMS.Purging.Strips.Three.Colors.zip)