bambulab/BambuStudio #5451: Another macos segfault

reedy, 2024-12-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.7.1

### Printer

P1S

### How to reproduce

Have BambuStudio open for an extended period, eventually, after sending to print, it'll segfault...

Seems to be a different segfault to https://github.com/bambulab/BambuStudio/issues/5053

### Actual results

Segfaults

### Expected results

Not segfaults

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/17971846/Archive.zip)
[BambuStudio-2024-12-02-020540.ips.zip](https://github.com/user-attachments/files/17971852/BambuStudio-2024-12-02-020540.ips.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

reedy, 2024-12-02:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [42425]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.10.01.50 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-02 02:05:36.5929 +0000
OS Version:            macOS 14.7.1 (23H222)
Report Version:        12
Anonymous UUID:        464EB762-6D2F-1499-C40A-5B51E69D4EF1

Time Awake Since Boot: 950000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace LIBSYSTEM, Code 2 Application Triggered Fault

Application Specific Information:
assertion failure: "!view->_hasCachedVisibleRect" -> %lld

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib                 0x185c01b4c __abort_with_payload + 8
1   libsystem_kernel.dylib                 0x185c27ed4 abort_with_payload_wrapper_internal + 104
2   libsystem_kernel.dylib                 0x185c27f08 abort_with_payload + 16
3   libsystem_c.dylib                      0x185b0d770 _os_crash_msg + 116
4   AppKit                                 0x18958f970 NSViewHierarchyInvalidateVisibleRect + 636
5   AppKit                                 0x18a01dc1c -[NSView _setHidden:setNeedsDisplay:] + 196
6   BambuStudio                            0x107905b64 wxWidgetCocoaImpl::SetVisibility(bool) + 28
7   BambuStudio                            0x107881dd4 wxWindow::Show(bool) + 100
8   BambuStudio                            0x10787a7f4 wxNonOwnedWindow::Show(bool) + 28
9   BambuStudio                            0x1078792f8 wxDialog::Show(bool) + 60
10  BambuStudio                            0x105a97878 Slic3r::GUI::SelectMachineDialog::Show(bool) + 348
11  BambuStudio                            0x107879510 wxDialog::EndModal(int) + 68
12  BambuStudio                            0x105632a48 Slic3r::GUI::Plater::print_job_finished(wxCommandEvent&) + 140
13  BambuStudio                            0x1079b9fe0 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
14  BambuStudio                            0x1079b9cc0 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
15  BambuStudio                            0x1079b9ae4 wxEvtHandler::ProcessEvent(wxEvent&) + 56
16  BambuStudio                            0x1079b9934 wxEvtHandler::ProcessPendingEvents() + 236
17  BambuStudio                            0x10791dcfc wxAppConsoleBase::ProcessPendingEvents() + 112
18  BambuStudio                            0x1079c27f0 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
19  CoreFoundation                         0x185d1485c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
20  CoreFoundation                         0x185d14748 __CFRunLoopDoObservers + 536
21  CoreFoundation                         0x185d13d74 __CFRunLoopRun + 776
22  CoreFoundation                         0x185d13414 CFRunLoopRunSpecific + 608
23  HIToolbox                              0x1904be19c RunCurrentEventLoopInMode + 292
24  HIToolbox                              0x1904bde54 ReceiveNextEventCommon + 260
25  HIToolbox                              0x1904bdd30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
26  AppKit                                 0x189572cc8 _DPSNextEvent + 660
27  AppKit                                 0x189d694d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
28  BambuStudio                            0x1078ea218 wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 180
29  BambuStudio                            0x1079c2cf8 wxCFEventLoop::DispatchTimeout(unsigned long) + 44
30  BambuStudio                            0x1079c2e18 wxCFEventLoop::OSXDoRun() + 172
31  BambuStudio                            0x1078ea6e8 wxModalEventLoop::OSXDoRun() + 88
32  BambuStudio                            0x1079c2ed4 wxCFEventLoop::DoRun() + 40
33  BambuStudio                            0x107942590 wxEventLoopBase::Run() + 132
34  BambuStudio                            0x1078793ec wxDialog::ShowModal() + 100
35  BambuStudio                            0x1055d0aa4 Slic3r::GUI::DPIAware<wxDialog>::ShowModal() + 124
36  BambuStudio                            0x1055e4ff8 Slic3r::GUI::Plater::priv::on_action_print_plate(Slic3r::GUI::SimpleEvent&) + 284
[... 621 lines cut ...]
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

Model: Mac14,3, BootROM 11881.41.5, proc 8:4:4 processors, 24 GB, SMC 
Graphics: Apple M2, Apple M2, Built-In
Display: DELL S2721DGF, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Apr  4 2024 20:56:02 version 23.30.58.0.41.51.138 FWID 01-60807d6e
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: Back-UPS RS 550G FW:857.L4 .I USB FW:L4
USB Device: USB2.1 Hub
USB Device: USB3.1 Hub
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.
```

walterwongbbl, 2024-12-12:
@reedy 
Thank you for providing the crash information. We will fix this issue