bambulab/BambuStudio #5454: Raspberry Pi 64 bit bookworm Does not install Network Plugin, Cannot use slicer in any way

rujulm-hub, 2024-12-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Raspberry PI bookworm 64 bit

### Additional system information

Broadcom BCM2711 quad-core Cortex-A72 (ARM v8) 64-bit

### Printer

Bambu A1 Combo

### How to reproduce

1. Launch
2. Popup: Downloading Bambu Network Plug-in
3. Shows: Install Successfully. 100%
4. Shows: Bambu Network plug-in not detected. Click here to install it.
5. This keeps repeating.

### Actual results

The Network-Plugin does not install and prevents me from finding my printer under the Device tab or even Export sliced files for SD card method.

Since the iPAD does not have a slicer installed I have no way to  use the printer.

### Expected results

Network Plug-In is installed properly.

### Project file & Debug log uploads

[debug_Sun_Dec_01_23_15_28_2.log.txt](https://github.com/user-attachments/files/17972955/debug_Sun_Dec_01_23_15_28_2.log.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-12-04:
hello @rujulm-hub 
I am afraid the network plugin currently is not supported on the linux system with arm chip,
if you want to use bambustudio on it, you need to export the .gcode.3mf into sd card, and plugin the sd card to machine

as the performance is limited, we suggest not to use bambustudio on such platform

autonumous, 2024-12-19:
Would be great to see support for ARM, especially the Raspberry PI.

RichColvin, 2025-01-12:
I am using a RPi 5 and having the same issue.

M1chlCZ, 2025-04-28:
Still have the same issue

ForTheRush, 2025-07-01:
same issue. no arm support is garbage in 2025