bambulab/BambuStudio #5458: Wrong color used

keycapss, 2024-12-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2

### Printer

Bambu Lab A1 with AMS Lite

### How to reproduce

1. Sliced model with full black body and white logo on the bottom, slicer showed that only black and white filament will be used.
2. I had one spool with about 50g of black filament so I started with it and had another black filament as a backup. All colors that were initially installed in AMS Lite: A1 White, A2 Silver, A3 Black (50g), A4 Black (1kg backup).
3. After the print started and my A3 slot did run out it swapped to A4 without issues. I put a new red filament into A3 slot because I would need it anyway for the next print.
4. After the print was about 60% done it for some unknown reason the printer unloaded black filament from A4 and loaded red filament from A3. Luckily I noticed that and manually unloaded all filaments except black that was in A4 slot and continued the print.

### Actual results

Printer would start printing with the wrong color even though in the slicer it showed only that white and black will be used. All filaments are from Bambu with RFID chip.

### Expected results

Printer would just finnish the print with black like it should have.

### Project file & Debug log uploads

[A1toolBOX.gcode.3mf.zip](https://github.com/user-attachments/files/17978933/A1toolBOX.gcode.3mf.zip)

https://makerworld.com/en/models/508656#profileId-424491

### Checklist of files to include

- [ ] Log file
- [X] Project file

gaolegao-lx, 2024-12-04:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

keycapss, 2024-12-04:
TECH_Unable to Feed Filament into the Extruder_EU_US241204344001_APP_ANDROID_3768 [ud:00822840]

keycapss, 2024-12-11:
Are you gonna fix this bug?