bambulab/BambuStudio #5459: 切片时没有生成支撑

deepzf, 2024-12-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Additional system information

amd 7840hs

### Printer

p1s

### How to reproduce

1、用0.2喷头的0.1层高标准配置，
2、打开树状支撑
3、调整模型摆放
4、切片

### Actual results

没有生成任何支撑

### Expected results

应该在大多数悬空墙上生成支撑

### Project file & Debug log uploads

[debug_Tue_Dec_03_00_05_57_6124.log.zip](https://github.com/user-attachments/files/17981013/debug_Tue_Dec_03_00_05_57_6124.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

deepzf, 2024-12-04:
不方便，使用0.2碰头配置，0.06和0.1的配置文件都有问题（其它没实验），树状支撑只有默认选项可以生成，苗条，有机，混合都不行

tangketan, 2025-02-21:
由于没有3mf，我只能大概猜一下原因。
支撑的悬垂角度阈值与层高有关。更小的层高意味着阈值角度也更小，也就是说，可以允许无支撑地打印更大程度的悬垂。
你可以尝试拖动修改角度阈值，看看是否生成了支撑。