bambulab/BambuStudio #5460: Unable to send files to sd card

insane78, 2024-12-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Send sliced file to SD (send only, not print)

### Actual results

I have 2 x1c printers in my lan, that are not configured as “lan mode” because i want use the app as well when i’m out of lab.
I have always been able to send the files to the SD card, but for some time I have no longer been able to, bambu studio ask me for the IP address and access code, entering the correct data, it asks me for the serial number, I enter it and nothing happens.
I would like to point out that the PC I use and the printers are in the same LAN and have the same IP branch (192.168.2.xxx)

### Expected results

File sent to SD 

### Project file & Debug log uploads

Problem is not about file, but networking...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ben-Pattinson, 2024-12-04:
I have the same issue. This is new to 01.10.01.50. Previously, the send was only stopped by the bug where you have a invisible character at the end of the code on the send screen (that you have to delete). Deleting that and clicking test used to then allow the send to continue. Whilst a pain and a bug, that was workable, I have no workaround for this one.

eastabrooka, 2024-12-10:
I am also seeing this.

Anthony-Bec, 2024-12-11:
I had this issue but I found out that I just needed to re login to by bambu account in studio

jarrard, 2024-12-14:
that does nothing for LAN MODE

walterwongbbl, 2025-01-01:
@insane78  @jarrard  @Anthony-Bec  @eastabrooka 
We apologize for the inconvenience. we will soon be using a new file transfer method, as FTP file transfers often fail due to network fluctuations.

insane78, 2025-01-01:
> @insane78 @jarrard @Anthony-Bec @eastabrooka We apologize for the inconvenience. we will soon be using a new file transfer method, as FTP file transfers often fail due to network fluctuations.

I never had any transfer problems before it stopped working, if the ftp module was the problem, it didn't work before either

StineDEk, 2025-01-05:
I have the same issue on my X1C

StineDEk, 2025-01-06:
Is there an ETA on this issue?

eastabrooka, 2025-01-19:
So... It looks like this breakage may be due to some **Real shady stuff** they are doing in the background. 

https://www.youtube.com/watch?v=aIyaDD8onIE

https://hackaday.com/2025/01/17/new-bambu-lab-firmware-update-adds-mandatory-authorization-control-system/ 

Like, maybe I'm jumping the gun here, but this looks super bad. 

I need a LAN Only mode.

thboehi, 2025-02-27:
> So... It looks like this breakage may be due to some **Real shady stuff** they are doing in the background.
> 
> Like, maybe I'm jumping the gun here, but this looks super bad.

This is going crazy... I was recommending this printer to everyone but now, that I have issues because of their cloud features, just to let me see my printer when I'm at school, this give me goosebumps..

I'll try to resell this printer and go for more open and less shady entreprise.