bambulab/BambuStudio #5462: The 3d-text module does not support Chinese fonts/创建3d文本功能不支持自定义中文字体

rgbpika, 2024-12-03:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 企业版 LTSC

### Additional system information

CPU: i7-10510U
MEMORY: 2*16GB 3200MHz

### Printer

Bambu Lab A1 mini Combo

### How to reproduce

1.以管理员权限安装中文字体给本计算机的所有用户
2.打开Bambu Studio中的3d文字创建功能
3.无法识别中文字体

### Actual results

1.无法识别中文字体
2.使用FontCreator将字体在中文环境下的名称中的中文内容去掉后，重新执行字体安装步骤，此时可以识别
3.识别后创建该字体下的3d文字，文字为默认字体，非所选字体

### Expected results

1.能正确识别中文字体
2.能正确创建该字体形状的文字

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17987749/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rgbpika, 2024-12-03:
[fontsFile.zip](https://github.com/user-attachments/files/17987872/fontsFile.zip)
字体文件及改名后的文件

Haidiye00, 2024-12-04:
Hello, we are currently resolving the font loading issue~

rgbpika, 2024-12-04:
> Hello, we are currently resolving the font loading issue~

Looking forward to your good news ~

Haidiye00, 2025-07-22:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~