bambulab/BambuStudio #5468: Screen function

3Dstitcher, 2024-12-04:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Home V23H2

### Additional system information

MSI
Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
16.0 GB

### Printer

Bambu X1-Carbon

### How to reproduce

Open Bambu Studio on laptop.
Maximize to the full screen. 
I can not access the taskbar in this mode, I have to minimize or use reduced size. 
I do use a secondary external screen, though HDMI connection, full screen mode on that functions correctly.
this is the only application/software that does this on my laptop.
This does not change when multi screen is disconnected

### Actual results

N/A

### Expected results

I assume this would need an update, unless there is a setting I'm unaware of, Thank you.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-12-06:
HI [3Dstitcher](https://github.com/3Dstitcher): 1. Could you pls provide a photo about the laptop that Maximize to the full screen。2.AndThe size of the notebook is how many inches, how much resolution Settings

bambu123, 2024-12-06:
Same As #2640

Awbmilne, 2025-05-12:
> Same As [#2640](https://github.com/bambulab/BambuStudio/issues/2640)
> […](#)

These seem to be different issues. #2640 handles the Auto-Hide taskbar, this (if I understand correctly) is an issue with the "maximize" window option covering a static taskbar.

I also have the later issue, where maximizing the windows covers the taskbar and I have to minimize or manually resize to access the taskbar.

The issue is not consistent. I just pulled up OBS to record the issue since it was occurring, and now it wont do it again. It seems to occur more often when switching between built-in display (laptop) and a multi-display dock or external HDMI monitor.