bambulab/BambuStudio #5470: 近期打开文件数量一多，顺序就开始混乱

walk728, 2024-12-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows11 23h2

### Printer

p1s

### How to reproduce

偏好设置里，近期项目的最大数量设置成199，然后正常使用，然后近期项目的最大数量超过100个左右以后

### Actual results

顺序就开始随机变乱，最近一次保存的是对的，永远在最上面，但是之前保存的顺序会乱。这个情况在100个以内的时候完全没有，全部都是按照打开顺序排列。

### Expected results

然后只要保存了一个项目，列表中的排序就会打乱

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18018334/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-12-06:
Yes, Now only two-digit numbers are supported.

bambu123, 2024-12-06:
现在只支持两位数的个数。