bambulab/BambuStudio #5474: All calibration buttons are grayed out

smwdscada, 2024-12-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu X1 Carbon

### How to reproduce

When I go to the calibration tab and select my printer, all the calibration buttons for Flow Dynamics and Flow Rate are grayed out.  My printer is online and connected.  I have performed these calibrations in the past so at some point they used to work.
[image]
[image]

### Actual results

Nothing happens when I click on the calibration buttons.

### Expected results

I expect to see the calibration wizard that walks the user through the process.

### Project file & Debug log uploads

[Bambu.zip](https://github.com/user-attachments/files/18027661/Bambu.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

hotellonely, 2024-12-13:
Yes it's still presenting. the only workaround is to open the tab as soon as you launch bambu studio.

SaltyAsFrig, 2024-12-18:
Same issue here too (Windows version). Can also confirm that hotellonely's workaround works too.

maruel, 2024-12-20:
See discussion at https://www.reddit.com/r/BambuLab/comments/1gpqypp.

R-A-S, 2024-12-23:
It has the same issue on Mac OS with the latest version of BambuStudio. It seems like all platforms have this issue.

zhimin-zeng-bambulab, 2024-12-27:
Thanks for reporting this issue, we  will fix it in the next version. You can reconnect the printer on the device page to update the status.

zhimin-zeng-bambulab, 2025-03-26:
It has been fixed in the latest version. You can have a try.

closed 2025-03-26 by zhimin-zeng-bambulab