bambulab/BambuStudio #5475: Replace with STL option is not working with STEP files

vadim-ryazanov, 2024-12-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Additional system information

Macbook Air M2 chip

### Printer

p1p

### How to reproduce

Load a project 
right click on an object and select "replace with STL option"
select a STEP file

### Actual results

original object is not replaced

### Expected results

object replaced with selected STEP file model

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-12-12:
same #5305

closed 2025-03-31 by MackBambu

ddmgh, 2025-08-24:
This issue appears to be back and I'm experiencing it with 2.2.0.85.