bambulab/BambuStudio #5477: On macOS, Cannot add a "pause point."

ToFuKing, 2024-12-06:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1.1

### Printer

Bambu P1S

### How to reproduce

On macOS, using the trackpad with two fingers to right-click

在 macOS 下, 触控板使用双指右键

### Actual results

you cannot add a "pause point."

无法添加”暂停点”

### Expected results

A right-click menu appears, allowing the addition of "pause points."

出现右键菜单, 允许添加"暂停点"

### Project file & Debug log uploads

no files

### Checklist of files to include

- [X] Log file
- [X] Project file

SamBenson, 2025-04-03:
Any update on this?

ToFuKing, 2025-04-06:
> Any update on this?

version 2.0.1.50, still no fix

LiZ-Li-BBL, 2025-04-17:
You can press the ctrl key while clicking the touchpad to simulate a right-click.

ToFuKing, 2025-04-17:
> You can press the ctrl key while clicking the touchpad to simulate a right-click.

It work!!!