bambulab/BambuStudio #5489: Text no longer edits

LTB3D, 2024-12-07:
### Bambu Studio Version

01.10.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia Version 15.1.1

### Printer

X1 Carbon

### How to reproduce

Attempt to edit the text.

1. Go to Objects
2. Select CHANGE TEXT
3. Select the text tool & Input Text
4. Change the words/letters to something else

### Actual results

Nothing happens. The text does note change & if you look hard enough, you can see the mouse curser flashing. 
This issue was reported my a user on MakerWorld. I tested & found the same results, but only after upgrading to this version.

### Expected results

The text type should change to the new input

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/18049737/Archive.zip)
[Logs.zip](https://github.com/user-attachments/files/18049738/Logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-12-10:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you

LTB3D, 2024-12-10:
[image]

Please see screen recording

Haidiye00, 2024-12-10:
[image]
Hello, can you provide this font? I noticed that I don't have it on my computer. thank you

LTB3D, 2024-12-10:
That's not the font that was used for this. It's not the first time that;s
shown up, but the font does not seem to matter. If you change the font,
nothing actually gets changed. It's like it gets stuck.
I went to another computer running an older version of the app & it edited
just fine. I upgraded that computer to the latest version & it now has that
problem too.

Haidiye00, 2024-12-10:
Hello, we have found that your operation has this issue with version 1.10.01.50. We have tested it with the latest internal version and it works fine. Please wait for the next version~

LTB3D, 2024-12-10:
Understood. Thank you.

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00