bambulab/BambuStudio #5490: 3D feild / render area not showing?

nyan-star1, 2024-12-07:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Printer

Bambu Lab P1S

### How to reproduce

1. Open Studio
2,Open new project
3.Check prepare and preview

### Actual results

[image]
[image]

### Expected results

The 3D feild is missing, even after a reinstall of the software, both flatpack and github

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18050205/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

nyan-star1, 2024-12-07:
(bambu-studio:9358): Gtk-CRITICAL **: 00:43:39.818: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

Keeps displaying this in the terminal, unsure how to fix it.

nyan-star1, 2024-12-08:
Issue was from being on Wayland without realizing, my bad.

closed 2024-12-08 by nyan-star1

Luky109, 2025-01-01:
i have the same issiue

Luky109, 2025-01-01:
[image]

Luky109, 2025-01-01:
can you please tell me how do you fix it??

nyan-star1, 2025-01-01:
> can you please tell me how do you fix it??
> 
> 

Yes! So! I was on Wayland, you need to use directX, when youre logging in you choose not to use Wayland, and it is fixed.

Luky109, 2025-01-01:
What is directX

Luky109, 2025-01-01:
I installed directx but i dont see it installed on my pc

nyan-star1, 2025-01-01:
Give me a minute, need to formulate myself.

nyan-star1, 2025-01-01:
Sorry, i mispoke, its X11, not DirectX, uninstall that.

X11 is a diffrent renderer.

What you need to do is, shut off your computer, when you log in, before clicking on your user theres a button in the bottom right, where you pick X11