bambulab/BambuStudio #5492: 软件切片直接闪退

ztdd2018, 2024-12-08:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Additional system information

Operating System: Windows 10 专业工作站版 64-bit (10.0, Build 19045) (19041.vb_release.191206-1406)
Language: Chinese (Simplified) (Regional Setting: Chinese (Simplified))
System Manufacturer: ASUS
Processor: Intel(R) Core(TM) i9-14900K (32 CPUs), ~3.2GHz
Memory: 131072MB RAM
Card name: NVIDIA GeForce RTX 3060
Manufacturer: NVIDIA
Chip type: NVIDIA GeForce RTX 3060
Display Memory: 77521 MB
Dedicated Memory: 12115 MB
Shared Memory: 65406 MB
Current Mode: 2560 x 1440 (32 bit) (59Hz)

### Printer

p1s

### How to reproduce

打开软件，导入模型，默认参数点击切片，闪退

### Actual results

软件闪退

### Expected results

软件闪退

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18051223/log.zip)
[rayquaza_mypokeprints+-+24.zip](https://github.com/user-attachments/files/18051225/rayquaza_mypokeprints%2B-%2B24.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-12-09:
HI [ztdd2018](https://github.com/ztdd2018)：Please try the following steps，https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

hotellonely, 2024-12-13:
the problem is that you have too much ram, which is a strange bug but plagues a lot of 64GB+ machines, same thing on orca as well.

QingZhangBambu, 2025-04-02:
[image]
这个dmp文件好像有点问题，无法打开
@ztdd2018 尝试一下最新版本，看看是否还有问题。如果仍然是崩溃闪退请上传新版本的dmp