bambulab/BambuStudio #5497: Multicolor not working when using CUT

JhunnelOrdinario, 2024-12-09:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu PS1

### How to reproduce

1. Use Cut in MultiColor Item
2. Preview but no color or single color only

### Actual results

The Item will generate single color and will ignore the original colors

### Expected results

Clicking Cut and Slice should keep the Color

### Project file & Debug log uploads

[Umbreon-Colored(2).zip](https://github.com/user-attachments/files/18061446/Umbreon-Colored.2.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-12-12:
Hello, currently our design is like this, and we will consider and make improvements in the future. Thank you for your feedback.

At present, we suggest that you cut first and then apply color to solve this type of problem~

stoilis, 2024-12-30:
I can confirm this bug.
In addition, the workaround of applying colour afterwards is not always possible.

In my current case, I have an elaborate model that I need to assemble and colour individually per component prior to the cut.
Due to the placement of each component in the final assembly, it's not possible to colour It after all of the pieces are put together.

ThorsReign, 2025-03-05:
> I can confirm this bug. In addition, the workaround of applying colour afterwards is not always possible.
>
> In my current case, I have an elaborate model that I need to assemble and colour [...]

Cut the model and assemble into objects with parts (right click and merge) then paint, then split to parts again.  The split keeps the painting.