bambulab/BambuStudio #5501: Print by color

BubbaTheMan, 2024-12-09:
I was printing something on my X1C with print by object and I had 2 different colors on 3 objects. One was grey the other two were blue. After one of the blue objects finished printing it moved to the grey object with a blue object still remaining.

I would like there to be a setting that makes it so that you can make it print my color.
what you want to happen.

An alternative for this would be just making it automatically print by color

[image]

mypillowsateyou, 2024-12-12:
you can rearrange the order of the objects in the list to set their print order

hadess, 2025-10-26:
> you can rearrange the order of the objects in the list to set their print order

That means printing by object and having more clearance between the 2 objects of the same colour than what would be needed to print by layer for each colour groups.