bambulab/BambuStudio #5504: Sending files to LAN X1 Carbon Bambu Timeout.

eastabrooka, 2024-12-10:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 

### Printer

X1 Carbon

### How to reproduce

Load Model. 

Slice.

Attempt to print to LAN 

### Actual results

Device does not print to LAN 
[image]

It times out. 

I went and inspected the logs. 
[Redacted Logs.txt](https://github.com/user-attachments/files/18081068/Redacted.Logs.txt)

2024-12-10 14:19:00.211894[Thread 0x00008fc0]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2024-12-10 14:19:00.213047[Thread 0x00008fc0]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
is repeated a lot. 

### Expected results

It should print as expected. 

### Project file & Debug log uploads

[Redacted Logs.txt](https://github.com/user-attachments/files/18081085/Redacted.Logs.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2025-01-01:
@eastabrooka 
We apologize for the inconvenience. we will soon be using a new file transfer method, as FTP file transfers often fail due to network fluctuations.

AlexEastabrook-NewburyInnovation, 2025-01-06:
Thanks for getting back to this issue, Look forward to getting a fix.

Let me know if there's a beta release out at any point with a bugfix. 

Currently manually loading files to the SD via USB.