bambulab/BambuStudio #5509: A1 with AMS prints differently than what I see on slicer

Urban82, 2024-12-10:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Manjaro Linux

### Printer

Bambu Lab A1 with AMS

### How to reproduce

1. Using model from https://makerworld.com/en/models/785772#profileId-723336
2. Changing color RED to GOLD
3. Slice and print with embedded settings

### Actual results

A GOLD ring is printed in the external part of the bowl in the same layer where the internal Santa face is printed

[image]

### Expected results

The bowl is printed without that external ring, as showed in BambuStudio (both "Prepare" and "Preview" pages):

Prepare:
[image]

Preview:
[image]

Preview (hiding the white filament):
[image]

### Project file & Debug log uploads

[Archivio.zip](https://github.com/user-attachments/files/18084605/Archivio.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Urban82, 2024-12-10:
Forgot to add:
Bambu Lab A1 FW 01.04.00.00
AMS lite FW 00.00.07.94

Urban82, 2024-12-11:
> Does it even attempt a color change to white at these layers or straight up just using gold ?

It regularly prints white where the white is in the preview, except for the external wall of those layers

gaolegao-lx, 2024-12-17:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

zhimin-zeng-bambulab, 2024-12-17:
Thanks for reporting this issue, we will check it