bambulab/BambuStudio #5511: Shadow model is not lifted by raft

UBaer21, 2024-12-11:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1S AMS×3

### How to reproduce

slice model with raft

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-12-14:
[image]

It seems a bug from Prusa, maybe it will be fixed with some time.

XunZhangBambu, 2024-12-29:
@UBaer21 The reason for this is that the raft is generated from data after slicing, whereas the shadow model is rendered directly based on the model data.

UBaer21, 2024-12-29:
As i had cases were the shadow made it harder to analyze (complex) layers - it would be nice to have an option to deactivate the shadow.