bambulab/BambuStudio #5516: Brim Ears [L] shortcut letter is incorrect

lrkaiser, 2024-12-11:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 10

### Additional system information

CPU: Intel(R) Core(TM) i7-4770S CPU @ 3.10GHz   3.10 GHz

### Printer

Bambu A1

### How to reproduce

1. Select an object
2. Type L or l

### Actual results

Opens support Painting (which is also [L])

### Expected results

Opens Brim Ears

### Project file & Debug log uploads

Open new project and add a primative.  Does this for any object.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-12-13:
confirm,it is already planned to change to B.

StdVectorBool, 2025-03-22:
> confirm,it is already planned to change to B.

Won't 'B' conflict with Mesh Boolean?
https://github.com/bambulab/BambuStudio/blob/4b48ad974389c3c20dfd605e9ddba0324787d27f/src/slic3r/GUI/Gizmos/GLGizmoMeshBoolean.cpp#L121

Haidiye00, 2025-03-23:
[image]
Hello, this is the latest update. Thank you for your discovery~

closed 2025-03-31 by MackBambu