bambulab/BambuStudio #5517: X-Y compensation per part instead of entire assembly

mypillowsateyou, 2024-12-12:
**Is your feature request related to a problem? Please describe.**
I added a surface Text part to an object, thus creating an assembly out of it. That object had some heavy X-Y contour compensation on it, -0.6mm. The Text part also gets this compensation and since it is so strong the text disappears completely after slicing, or with detect thin walls looks bad

**Describe the solution you'd like**
Would like to be able to define separate X-Y contour and hole compensation per part in an assembly. That way the text could stay the normal size while the rest of the object where tolerance matters can be adjusted

**Describe alternatives you've considered**
- Hand write or mark the tolerance after. Could be a problem since I am printing many test copies of the same thing with different tolerances, with the text identifying the tolerance so I don't get parts mixed up coming right off the build plate
- Thicken the text so it is legible even at high compensations. Problem is the parts are kind of small and can't make the text big enough to survive the compensation. Also would be nice if the text looked similar style between parts
- Use color code per part for tolerance. Limited by AMS count, also not quickly identifiable, need to translate color to tolerance
- Add a large feature to each part protruding from top which won't disappear with compensation and which has height same as tolerance of that part and can be measured with calipers. Would be a pain
- Split assembly to separate objects, the original object and text object and use different X-Y comp for them. Since an object can't float above the build plate, it would have to be printed at the base of the part ideally with a different colored filament which requires AMS. This is probably what I will do instead for now

**Additional context**
unsliced
[image]

sliced
[image]

Lukrative525, 2025-01-11:
I would also be interested in per-object x-y contour compensation. In my case, I have a part that I modeled the support for. I'd like to adjust the x-y contour compensation of the modeled support to be slightly negative so that it just barely doesn't touch the part wall loops. See the images below:

Overall Project:
[image]
[image]

Project at a given layer:
[image]

Red indicates where a gap is desired (by adjusting the x-y contour compensation of only the support):
[image]

Lukrative525, 2025-01-17:
It seems that this is already an option for individual objects, but not for individual parts of an object. Bummer.