bambulab/BambuStudio #5523: Erroneous Server Exception Alert

tominco, 2024-12-12:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.1.1

### Additional system information

CPU: Apple M4
Memory: 24GB
Graphics: Apple M4 Built-in GPU

### Printer

Bambu P1S

### How to reproduce

After slicing model, simply click Print Plate

### Actual results

The model prints as expected; however, a dialog pops up that says
[image]

It happens every time and has been for several weeks. At least since the last Bambu Studio update maybe longer. Clicking the Don't Show this dialog again box has no effect going forward. It still shows up.

There is no indication that there are any problems with current system services when I check.

### Expected results

I should not get that dialog.

### Project file & Debug log uploads

[Part Studio 1 - Part 2.3mf.zip](https://github.com/user-attachments/files/18117383/Part.Studio.1.-.Part.2.3mf.zip)
[debug_Thu_Dec_12_13_40_42_76073.log.0.zip](https://github.com/user-attachments/files/18117407/debug_Thu_Dec_12_13_40_42_76073.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

dts350z, 2024-12-13:
FYI I'm getting this consistently on windows 11 23H2.

Nanobot1309, 2024-12-27:
Same here several weeks
I have here 2 X1c printers and when i start BambuStudio and have the same problem.
Don’t show this dialog again no effect.
it happens every time i start bambustudio.
tried many things: firewallsettings windows 10 , uninstall en reinstal bambustudio , clean settup off network , log out/in bambustudio.
Nothing helps.

Hope Bambulab fix it very soon , getting pretty sick of it.

[image]

philyg, 2025-01-02:
Same on Windows 10 with a P1P, Bambu Studio version 1.10.1.50. I havent been able to find _any_ other networking problems. Seems to me like it's some misconstructed request on the client software side or some mishandled request on the bambu server side (or an expected error that isn't correctly ignored on the client side).

Nanobot1309, 2025-01-03:
Here Windows10 with latest bambustudio,
Tried an older verrsion of bambustudio and no problems.

grz Jeroen

walterwongbbl, 2025-01-08:
Hi. @tominco @philyg @dts350z 
I'm sorry to hear that you're experiencing 404 or 500+ errors with certain services. I will review the logs you provided, and for some non-critical errors, I will refrain from displaying this dialog box again. Thank you for your understanding.

philyg, 2025-01-11:
Problem persists in beta 1.10.2.64, plugin version 1.10.02.06, build 20250109-113132, europe region.

burgbach, 2025-01-13:
Same issue on MacOS 14.7, Bambu Studio 01.10.01.50
Using the included network test, I see two failing tests: storage and storage upgrade. The first one logs this:
Mon Jan 13 11:07:05:test storage start…
Mon Jan 13 11:07:05:[test_oss]: url=http://upload-file.bambulab.com
Mon Jan 13 11:07:06:test storage failed
Mon Jan 13 11:07:06:status=404, body={« error_msg »: »404 Route Not Found »}
, error=

Second one:
Mon Jan 13 11:08:02:test storage upgrade start…
Mon Jan 13 11:08:02:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Mon Jan 13 11:08:02:test storage upgrade failed
Mon Jan 13 11:08:02:status=404, body={« error_msg »: »404 Route Not Found »}
, error=

dgarcia-1220, 2025-01-14:
This seems to happen to me every time Bambu Studio is trying to request the rating for the previous print profile. After I finish a print successfully, switching to the "Device" tab will cause this error popup every time until the next print is started. This has caused me to stop receiving boost tokens for successful prints.

philyg, 2025-01-16:
Problem persists in beta 1.10.2.64, plugin version 1.10.02.09, build 20250109-113132, europe region.

zbrozek, 2025-01-19:
This continues to be a nuisance. Is there any way to escalate?

tominco, 2025-02-11:
It's been nearly two months, is anyone going to look at this and try and fix it? It's really annoying. I will see it as many as four times from slicing to printing in a 30 second time frame.  If nothing else make the "don't show this again" check box work.

tominco, 2025-02-14:
Update.. Several times now, when the error dialog has appeared, BambuStudio becomes unresponsive. I can't close the dialog, or even quit the app. I'm ultimately have to use Force Quit (Mac OS) to terminate the BambuStudio process and start over again.

internoot, 2025-02-15:
This error started occuring for me when I enabled 'incognito mode' in my makerworld account settings.

Is this unique to me or are others also using incognito mode?

tominco, 2025-02-15:
I have been using Incognito mode since day one with my printer (March 2024). The problem didn't start until December 2024.

dgarcia-1220, 2025-02-20:
Disabling incognito mode resolved this issue for me.