bambulab/BambuStudio #5527: Window configuration/graphics error?

ihillway, 2024-12-13:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

6.11.10-304.bazzite.fc41.x86_64 (64-bit)

### Additional system information

16 × AMD Ryzen 7 5800X 8-Core Processor
NVIDIA GeForce RTX 3070 Ti/PCIe/SSE2

### Printer

P1S with AMS

### How to reproduce

[image]
[image]
[image]
Issue is present as soon as you open the software. The print options tabs on the left appear to be missing content and you cannot scroll down. The model preview area is completely blank. Models that are already ready to go can be printed just fine though. Functionality seems to be intact. I suspect this is a rendering issue.

### Actual results

There seems to be full functionality despite the user interface problems.

### Expected results

You should see the plate area preview and model placement. Areas on the left should have scroll bars and more content.

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/18120250/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ihillway, 2024-12-13:
Forgot to mention, downloaded  through Discover as flatpak.

MackBambu, 2024-12-23:
same issue, try reverting the NVIDIA driver to version 555.
https://github.com/bambulab/BambuStudio/issues/4669