bambulab/BambuStudio #5530: Limiting platform peak power draw at over 220V

mildw4ve, 2024-12-13:
**Is your feature request related to a problem? Please describe.**

Bambu P1, X1 and A1 can draw over 1100W of power at a voltage of 220V or higher and only 350W at 110V. This is of course mainly influenced by AC platform heater. It creates a problem for people in Europe for example where 220-240V i standard as it limits the number of machines one can run concurrently without major electrical rework.

**Describe the solution you'd like**

An option, setting within the slicer or the machine itself to limit maximum power draw of the platform. Open source firmwares such as klipper or reprapfirmware allow for heater power limiting.

**Describe alternatives you've considered**

- Custom platform heating gcode, setting temperature targets in small increments - I tested it but still saw spikes up to 900W
- External 220>110V transformer - big, noisy, expensive.
- Printer hardware modification.

larsgoolsen, 2025-12-01:
I know it’s been a year, but I just made two quick prints over the last hours, while my A1 was attached to my 2kW powerstation, and at NO time was it above 380W. This is the graph:

[image]

larsgoolsen, 2025-12-01:
I printed PLA for this test, and I’m on 230v (it’s practically all we have in Denmark). Once I get my hands on other filaments I’ll probably update this thread.
The powerstation is pretty accurate (already verified that) but the graph is not as granular as the live-portion of the app, so I followed that during warm-up phase.
Are you on latest firmware? They might have adjusted things?