bambulab/BambuStudio #5531: AMS Auto Refill interprets "No Color" as a distinct color for the purposes of switching spools during print

re3dCharlotte, 2024-12-13:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel(R) Xeon(R) E-2176M  CPU @ 2.70GHz   2.71 GHz

### Printer

Bambu Lab A1 with AMS Lite

### How to reproduce

Using non-bambu filament on AMS, manually set AMS filament type to generic PLA for all AMS slots. Do not select a color. Have Auto Refill enabled. run a print where one spool on AMS runs out during print.

### Actual results

The printer will automatically switch to a different spool on AMS lite during the print.

### Expected results

Not selecting a color should not allow auto refill to work. ie: the default setting on AMS is no color, if the user has not set the color, the default behavior from a lack of information should not be to interpret 4 no color generic pla slots as the same color and therefore switch on runout. This was the issue I accidentally discovered today. I expected the printer to pause when it ran out of the spool and wait for a refill, but it instead auto refilled from an unintended slot.

i think there should be a coded exception to require a color to be input in order to allow for auto refill.

### Project file & Debug log uploads

[Articulated Gingerbread Dragon Fully Colored.zip](https://github.com/user-attachments/files/18129229/Articulated.Gingerbread.Dragon.Fully.Colored.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MartinChan3, 2024-12-14:
According to your description, this issue seems to be a feature rather than a bug.

re3dCharlotte, 2024-12-16:
> According to your description, this issue seems to be a feature rather than a bug.

I disagree because all documentation refers to AMS color switching requiring same brand and same color in order to switch, and therefore a user who had not input a color choice would not expect the colors to switch.