bambulab/BambuStudio #5532: Live view seems to be a lot worse then in the previous version

mrootje, 2024-12-13:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

13th Gen Intel(R) Core(TM) i7-1365U   1.80 GHz, 32 GB RAM

### Printer

X1C

### How to reproduce

Go to "device" tab and click on play.

### Actual results

Very, very stuttery image, quite useless.

### Expected results

Smooth video like it used to be. Seems to be a bambu Cloud problem: my network devices are largly idle as is my internet connection (1Gbps) . 

### Project file & Debug log uploads

Generic problem not related to a specific project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mrootje, 2024-12-13:
Not project related.

bambu123, 2024-12-18:
Can you provide a video of the problem so that we can confirm the specific issue?

Sortam, 2025-01-28:
All video significantly worse now. The only difference I see is a much lower bitrate.

https://github.com/user-attachments/assets/56394a9c-7a7c-4052-a9dc-17eb5070784e

[image]