bambulab/BambuStudio #5534: Plate type changed when reopening project.

blaskotron, 2024-12-13:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Set build plate to cool type (project 1)
2. Save project
3. Close project
4. Start new project (project 2) 
5. Change build plate to textured pei plate. 
6. Close project 2 
7. Reopen project 1. 

### Actual results

Build plate that was saved in project 1 has been changed to textured pei plate instead of cool plate. 

### Expected results

Expect project 1 to have build plate type that was saved. I just clogged my nozzle by bambu changing the plate type without warning. 

### Project file & Debug log uploads

[cool plate.zip](https://github.com/user-attachments/files/18131695/cool.plate.zip)
[log.zip](https://github.com/user-attachments/files/18131699/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MartinChan3, 2024-12-14:
@blaskotron No, I don’t think it’s a bug. This is because for most customers, changing the print bed is not a frequent operation. As for the 3MF format, it might store various types of print beds, which means that downloading a new 3MF from scenarios like MW could almost always result in a risk of print failure.

lanewei120, 2024-12-18:
[image]

Hello @blaskotron 
for your use case, you can turn off auto plate type in preference window
it will use the one saved in 3mf

thanks