bambulab/BambuStudio #5539: Unable to Scroll through models - Online Models Tab

Androfred, 2024-12-14:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

CPU: Ryzen 9 7950X
Memory - 128GB DDR5
GPU - RTX 3090 24GB
**4K Display (3840x2160 - 100% Scaling)** (Important to reproduce)

### Printer

X1C

### How to reproduce

1. Open Bambu Studio on a high resolution display
2. Open the online models tab and type in any search feature that has more than 20 results
3. Attempt to Scroll

### Actual results

20 Results will show, but you will be unable to scroll your mouse wheel in order to view more

[image]

### Expected results

The expected result would be to scroll further, and have it continue pulling up relevant files/prints related to the search results.

Workaround: (making the window smaller so that the searches occupy all the way "past" the screen yields the expected results until a new search request is entered. A potential fix could be to load 30-40 results to start instead of 20. 

[image]
[image]

### Project file & Debug log uploads

No Project file required to reproduce bug
No Log File generated as this is not a crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GerHar, 2024-12-18:
Hi, I think is related that for Ubuntu 24.04 the search for the same home tab is not allowing to type nor search. Is this being tracked somewhere else? Thanks!

bambu123, 2024-12-18:
We will optimize it, thank you

realzorro, 2024-12-18:
The cause of the problem has been identified and a version fix will be released next week.
Thank you for your feedback, thank you.

GerHar, 2024-12-26:
You are awesome!