bambulab/BambuStudio #5541: cannot build on mac

stewartoallen, 2024-12-14:
### Bambu Studio Version

git HEAD / master

### Where is the application from?

Bambu Lab github releases

### OS version

OS X 10.4

### Additional system information

Mac Studio M2 Ultra

### Printer

build process

### How to reproduce

following the guide at https://github.com/bambulab/BambuStudio/wiki/Mac-Compile-Guide

### Actual results

after deps are build, attempting to build BambuStudio yields:

```
CMake Error at CMakeLists.txt:2 (project):
  Generator

    Xcode

  does not support platform specification, but platform

    arm64

  was specified.
```

### Expected results

successful build

### Project file & Debug log uploads

see above

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-12-23:
Can you provide the complete log? I'm not sure if there are other issues.

stewartoallen, 2024-12-26:
there are no other logs. I created and populated the deps and issued the cmake command and that's the entirety of the output. it never started building, just provided that error. here's the command. `$BSD` is an env to the deps directory. same results if I put in the full dir path. this is just shorter to paste:

```
cmake ..  -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="$BSD/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH="$BSD/usr/local" -DCMAKE_MACOSX_BUNDLE=on
```

MackBambu, 2024-12-30:
I successfully built using the following command on macOS 12.6 (the minimum version I tested), and the build also works correctly on macOS 13 using GitHub Actions. However, I haven't tested it on macOS 10. You can try running the command below or check if your CMake version is too old.
```
cmake .. -DCMAKE_PREFIX_PATH="$BSD/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH="$BSD/usr/local" -DCMAKE_OSX_ARCHITECTURES="arm64" -DCMAKE_MACOSX_BUNDLE=ON -DBBL_RELEASE_TO_PUBLIC=1 -DBBL_INTERNAL_TESTING=0
```

stewartoallen, 2025-01-05:
are you building on an Arm or Intel Mac? Cmake is up to date.

MackBambu, 2025-01-11:
> are you building on an Arm or Intel Mac? Cmake is up to date.

arm mac

stewartoallen, 2025-01-12:
idk then .. same setup here

MackBambu, 2025-01-12:
Try updating your system version or toolchain versions such as CMake and Clang. It's hard to guarantee that any version will build successfully.

stewartoallen, 2025-01-13:
it's a brand new mac mini m4 and I just installed xcode, brew, and all the deps. could not be more current.

MackBambu, 2025-03-12:
macOS sdk 14 and 15 is not yet supported for compilation. There are still many unresolved compilation errors.
https://github.com/bambulab/BambuStudio/pull/3968