bambulab/BambuStudio #5543: 布尔运算在部分模型上不起作用

moe001, 2024-12-14:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Additional system information

10940x/32g/3090

### Printer

p1s

### How to reproduce

1. 打开附件工程
2. 点击盘上的模型
3. 可选：修复模型，以确定（本软件认为）模型没有非流形边问题
4. 选择菜单上的布尔运算
5. 选择差集
6. 先点击弯月状的模型(part_1...)，再点击内部的类三角形模型(part_13)，勾选“删除输入”并确认
（也就是在前者上删除这个三角形并且也删除三角形本身）

### Actual results

“part_1...”模型上没有被正确的开出三角形的孔洞，其应被完整切割的尖角，布尔之后仍然是一个诡异的三维平角，参见图片。
[image]

### Expected results

正确的切割模型，上图中绿色虚线包含的部分，应被移除。

### Project file & Debug log uploads

[bug.zip](https://github.com/user-attachments/files/18138105/bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2025-01-06:
[image]
Hello, the latest internal version will generate errors~

Haidiye00, 2025-01-11:
Hello, the new beta version has been released. Welcome to download and experience it~
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-11 by Haidiye00