bambulab/BambuStudio #5544: Text no Longer goes on Domed surface

geb63619, 2024-12-15:
### Bambu Studio Version

1.10.00.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Device name	Basement
Processor	Intel(R) Core(TM) i3-N305   1.80 GHz
Installed RAM	8.00 GB (7.66 GB usable)
Device ID	5A392F22-A55A-43B1-97A0-54ED3DAEA713
Product ID	00342-21126-02249-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Bambu Lab X1C

### How to reproduce

I have a doomed clay pigeon I put text on that no longer snaps the text to the doomed area of the stl file when I go to change the name.  

### Actual results

New version no longer snap text to the domed area when trying to change the name. 

### Expected results

Can we get this fixed I used this feature all the time.  Currently went back to version 1.9.2.57 where this feature works

### Project file & Debug log uploads

If you open this file in an older version you can change the name and it snaps to the surface in the new version it no longer snaps to the domed surface.

[Bambu Issue.zip](https://github.com/user-attachments/files/18138263/Bambu.Issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-12-18:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1

Haidiye00, 2025-01-13:
[image]

closed 2025-01-13 by Haidiye00

geb63619, 2025-01-13:
> Hello, the new beta version has been released. Welcome to download and experience it~
> https://github.com/bambulab/BambuStudio/releases

Thanks I will download and try the new version 
Gary

geb63619, 2025-01-13:
Thanks
I'll get it downloaded and try it out thanks so much
Gary

 
 

https://github.com/bambulab/BambuStudio/releases