bambulab/BambuStudio #5545: Error slicing

Keving1950, 2024-12-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

windows 11 on LG Gram laptop

### Printer

A1

### How to reproduce

added svg file to base, prepare looks OK
When I slice logo does not show

### Actual results

1. Preview looks ok have SVG logo on badge plate
2. slice
3. logo does not show no error
[image]
[image]
[image]
 messages

NO error messages

### Expected results

print as in prepare tab

### Project file & Debug log uploads

[debug_network_Sat_Dec_14_09_38_43.log.zip](https://github.com/user-attachments/files/18138262/debug_network_Sat_Dec_14_09_38_43.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2025-01-06:
[image]
@Keving1950  It may be because the line width of this SVG logo is too small, making it impossible to generate two layers of walls through slicing.
1. You can set the wall generator to Arachne mode as shown above picture
2. Could you share your 3mf file?

Keving1950, 2025-01-06:
Thanks
Tried Arachne, no different
been playing a lot and succeeded using designspark to create new badge,
both new and faulty bagge attached
Kevin Guest
0412874400

zhimin-zeng-bambulab, 2025-01-06:
@Keving1950  Could you share your 3mf file to better locate the cause of the problem?

Keving1950, 2025-01-06:
attached last email
Kevin Guest
0412874400

zhimin-zeng-bambulab, 2025-01-06:
[image]
Sorry, I can't open the connection you provided.

Keving1950, 2025-01-06:
> image.png (view on web)
> Sorry, I can't open the connection you provided.

how do I send file
Kevin Guest
0412874400

zhimin-zeng-bambulab, 2025-02-05:
@Keving1950  You can compress this file into zip format and upload it, or send it to my email (zhimin.zeng@bambulab.com). Sorry for the late reply.