bambulab/BambuStudio #5546: No AMS filaments. Please select a printer in 'Device' page to load

KarlGrear, 2024-12-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1.  Open Bambu Studio
2. Click Sync Filaments with AMS
3. Get "No AMS filaments. Please select a printer in 'Device' page to load "
4. Click on the device tab
5. Click back to Prepare tab
6. Click Sync Filaments
7. Filaments sync as expected.
8. Load project
9. Click on Sync filaments
10.  Get "No AMS filaments. Please select a printer in 'Device' page to load "
11. Click on the Device tab.
12. Click back to prepare
13. Click on Sync Filaments
14. Filaments sync as expected.

This issue just started today.
Tried to reboot both the printers and my computer.
Tried logging out and back into the printers.
I have 2 X1C printers, it does not matter which is selected in the device tab.
It happens with no project or any project that I load, including ones that I previously printed without issue. 

### Actual results

The above explains the problem and steps to reporduce

### Expected results

Would sync the ams without having to visit the device tab.

### Project file & Debug log uploads

[debug_Sat_Dec_14_21_11_48_38628.log](https://github.com/user-attachments/files/18138918/debug_Sat_Dec_14_21_11_48_38628.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DPattee, 2024-12-15:
I encountered this one time yesterday and one time today. Same studio version as listed above. But, on macos 14.7. 

I did multiple prints of different projects, and it did not reproduce every time.

hcancelik, 2024-12-19:
This happens to me every time I open bambu studio as well. In fact I cannot even select a filament unless I go to device menu. This wasn't an issue when I only had A1 mini. Right after I added a second printer, I started to have this issue.

xanomanox, 2025-01-08:
I started getting this problem in December, P1S. Issues started around the time I got TPU for AMS and did a firmware refresh so the TPU spool could be recognized. Now this.

DanBao-Bambu, 2025-01-17:
The issue has been recorded in the requirement list.

JohnSmithCYM, 2025-02-24:
@KarlGrear We have tried to solve this problem with 1.10, you can download the new version(https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.72) to check. If you can reproduce this problem, please let us know.

hcancelik, 2025-02-24:
@JohnSmithCYM I just tried the new beta version and the issue still persists.